jmcerrejon/PiKISS #3: Emulators doesn't seems to work

asfo, 2019-01-06:
Hope you can see my issue :)

I tried to install PiKISS but doesn't seems to work on Raspberry Pi Zero (W)...

I did the git clone, and executed the .sh file so everything went fine there, but after I install the GBA Emulator (Or SNES, I just tried those) shows an error that the "helper.sh" can't be found...

After that shows that the emulator "is installed correctly" but when I press "Y" to start the default game/free game or something like that to test it, it does nothing because an error (I don't have my RBPZ right now to show you but if you give me an answer here I will upload an screenshot)

jmcerrejon, 2019-01-15:
Didn't have a Raspberry Pi Zero to test. I'll try to test with RPi 3 Model B+ (2017)

jmcerrejon, 2019-01-15:
Don't have a Raspberry Pi Zero to test. I'll try to test with RPi 3 Model B+ (2017)

actraiser30, 2020-02-20:
Hello,can you make a script to be able to install PiKiss to the Retropie ??

or it is possible like this if yes how?

thanks

jmcerrejon, 2020-02-20:
> Hello,can you make a script to be able to install PiKiss to the Retropie ??
> 
> or it is possible like this if yes how?
> 

_Retropie_ is a _Debian_ distro, so login into the OS through _SSH_ and clone this repo. Piece of cake! 😊

actraiser30, 2020-02-20:
thanks what i mean is to make a script for the retropie setup menu to install it from there like installing a port for example

actraiser30, 2020-02-20:
Also can you put in the server menu the software Transmission torrent server?

something else,the ramlog page does not exist anymore so you cannot install it, even if you could install it is way outdated and as reported by many people not working correctly,you can replace it with the log2ram though:

https://github.com/azlux/log2ram

jmcerrejon, 2020-02-20:
> thanks what i mean is to make a script for the retropie setup menu to install it from there like installing a port for example

Oh, I know your thoughts. Never build a script for RetroPie. I note it down in my TODO list with a 
 script to install a torrent server. 

> Also can you put in the server menu the software Transmission torrent server?
>
> something else,the ramlog page does not exist anymore so you cannot install it, even if you could install it is way [...]

Thks for the help with ramlog. Didn't know that.

actraiser30, 2020-02-20:
if you can put Transmission torrent server in servers will be great thanks

actraiser30, 2020-03-03:
Hello, in dune legacy the mouse does not work only the keyboard works is there a setting or workaround to correct this?

jmcerrejon, 2020-03-05:
Let me check it out this weekend.

jmcerrejon, 2020-03-06:
I use to support the latest Raspbian version. The game is playable and it works flawlessly on Raspbian Buster.

I guest your problem is with SDL2 lib on Stretch. Try another SDL2 game from Debian repositories. You have another script to install manually SDL2, follow the script step by step. My recommendation is that upgrade your Raspbian  to Buster 😄

Good luck!

closed 2020-03-06 by jmcerrejon

actraiser30, 2020-03-06:
i tried it in both buster and stretch in buster it was working but the mouse pointer is not visible do you know how to fix this?

jmcerrejon, 2020-03-06:
like I said, your issue is with SDL2, I just tested with the latest Raspbian and all is working.

actraiser30, 2020-03-06:
how did you install sdl exactly?

jmcerrejon, 2020-03-06:
It’s already installed

actraiser30, 2020-03-06:
dune legacy does not install sdl2 if i run the game like this i get an sdl error

actraiser30, 2020-03-06:
anyway thanks i will figure this later for now it will be great to add the transmission torrent server and the plex media server.

and more games in the future if you need help let me know