jmcerrejon/PiKISS #12: Error message on rpiOS

Mikez333, 2020-07-08:
I'm on the latest version of rpiOS, and I got `ping: socket: Operation not permitted
Internet connection required. Check your network.
`
I'm pretty sure I have internet, or else VNC Viewer would not connect...

jmcerrejon, 2020-07-08:
Did you found this issue when type or click on piKiss or in a particular script?

Mikez333, 2020-07-08:
Just clicking or running the script. When I click actually, The terminal windows opens and disappears. Also, there is no pikiss button in the application menu or a system tools menu.

jmcerrejon, 2020-07-09:
After our conversation through Discord, just trying you environment with a Raspberry Pi 3 and WiFi. It's working here.

[image]

Check the next:

1) Open a terminal and type:

```
cd piKiss
./piKiss.sh -ni
```

2) Open a terminal and type:

`echo ping -c 1 8.8.8.8 > /dev/null && echo 'true'`

It should be **true**

Mikez333, 2020-07-09:
Ayy it worked! I also just tried this on my pi 4, and it worked too. Looks like the .desktop file needs to be changed to /home/pi/piKiss/piKiss.sh -ni to be able to open pikiss. You should probably add that to the .desktop file code in the repository. Thx!

Roburetto, 2020-08-04:
For the record: was having this issue on Twister OS 1.4.1 (which ships PiKiss by default).

It turns out same distros do not let unprivileged users use ping by default.

This fixed it:  
`sudo setcap 'cap_net_admin,cap_net_raw+ep' $(which ping)`

(from https://github.com/MichaIng/DietPi/issues/1012#issuecomment-569012816)

mobilegmYT, 2020-08-08:
I ran it any still got 
`Internet connection required. Causes:

 · Check your network.
 · Weak WiFi signal?.
 · Try no check internet connection parameter (-ni): cd ~/piKiss && ./piKiss.sh -ni

Press [Enter] to exit...
`

jmcerrejon, 2020-08-08:
Are you sure you have good WiFi signal?. Did you try wired connection?. It works here.

mobilegmYT, 2020-08-08:
I have wifi cuz I'm connecting to my pi via VNC.

jmcerrejon, 2020-09-13:
I didn't receive any issue about this problem again. I'll close this issue.

closed 2020-09-13 by jmcerrejon