jmcerrejon/PiKISS #13: get Games to work on Pi4

Gert1967, 2020-07-11:
installed the 64bit beta PiOs on my Pi4 with Pikiss and dukenukem and supermario from your games section. when i start the games i get the following message:

./sm64: error while loading shared libraries: libGLESv2.so.2: cannot open shared object file: No such file or directory

can you give me further assitance to get the games working. thanx in advance - gert

jmcerrejon, 2020-07-11:
Not 64 bit compliance. 64 bit it's in a very early stage, but maybe you could compile with 64 bit support.

Gert1967, 2020-07-11:
ok ... should read before installing:
2) libraspberrypi0, libraspberrypi-dev and libraspberrypi-doc have been moved out of /opt/vc/* and into /usr/* instead (making it more standard). Any code built against these libraries will require changing to refer to a more standard location (/usr/lib/ rather than /opt/vc/lib)
3) Due to 2) Many packages that expect libGLESv2.so libEGL etc will require rebuilding.

jmcerrejon, 2020-07-11:
You are right. I talked about it with Salva (Pi Labs) and the idea is install in a future all libraries in a more standard stage.

Gert1967, 2020-07-12:
gave the 32 bit os a try and everything works fine ...

closed 2020-07-12 by Gert1967