jmcerrejon/PiKISS #20: REQUEST: remove vulkan drivers?

puccaso, 2020-08-01:
is there anyway to add a undo/remove vulkan drivers/roll back feature?
it works great in windows, but when full screen, the screen tears and becomes jumbled really badly.

jmcerrejon, 2020-08-01:
Sorry. Not possible atm. 😅

Roburetto, 2020-08-02:
Vulkan seems to be working in fullscreen with PPSSPP here.

A question: is there a problem if we delete the leftover /pi/mesa_vulkan folder AFTER the driver is installed?

jmcerrejon, 2020-08-02:
Delete it if it's your desire and if something goes wrong, install it again with PiKISS 😉

puccaso, 2020-08-05:
i was able to fix the tearing by changing the graphics from Fake KMS to the Full KMS by adding the -pi4 to the overlay..
all good now 👍 
thank you guys!

closed 2020-08-05 by puccaso