jmcerrejon/PiKISS #21: REQUEST: could you pls add supertuxkart

skullphantom241, 2020-08-04:
i have tried installing supertuxkart through snapd in terminal but it wont work. could you add supertuxkart into pikiss if possible

Roburetto, 2020-08-04:
supertuxkart package in Raspberry OS repo is version 0.9.3. A bit old by now.

Also it complains about OpenGL being used (v2) being too old. As far as I understand, more recently fkms driver supports OpenGL v3.

Would be great to have an update.

Blueyzachary, 2020-08-04:
sudo apt update && sudo apt install supertuxkart -y

-Zach Smith
blueyzachary@icloud.com
(801) 400-6238

Roburetto, 2020-08-04:
Thanks for the info. 

I see there is a new version here: http://archive.raspbian.org/raspbian/pool/main/s/supertuxkart/

But the command you suggest is still returning:
"supertuxkart is already the newest version (0.9.3-2).""

my: /etc/apt/sources.list;
deb http://raspbian.raspberrypi.org/raspbian/ buster main contrib non-free rpi

jmcerrejon, 2020-08-04:
Maybe you can manually download the .deb and install it. Don't forget the data file.

Roburetto, 2020-08-04:
> Maybe you can manually download the .deb and install it. Don't forget the data file.

No luck with a manual download & install. Met dependency issues (seems the build uses libs newer than available in stable distro ?).

Os pacotes a seguir têm dependências não satisfeitas:
 supertuxkart : Depende: libc6 (>= 2.29) mas 2.28-10+rpi1 está para ser instalado
                Depende: libgcc-s1 (>= 3.5) mas não é instalável
                Depende: libnettle8 mas não é instalável
                Depende: libstdc++6 (>= 9) mas 8.3.0-6+rpi1 está para ser instalado
E: Não foi possível corrigir problemas, você tem pacotes mantidos (hold) estragados.

closed 2020-09-13 by jmcerrejon