jmcerrejon/PiKISS #22: DELETED MY HOME FOLDER DURING UNINSTALL

mobilegmYT, 2020-08-26:
I have nothing to say. 
[image]
All files in pi were deleted and maybe other places were affected?

jmcerrejon, 2020-08-27:
Apologies. from where have you run PiKISS?. Menu icon, created Desktop icon, /home/pi/piKiss/piKiss.sh, cd /home/pi/piKiss && ./piKiss.sh

jmcerrejon, 2020-08-27:
I tried all possible stages and the only way to get this was to invoke the script from a directory that was not _piKISS_, like:

```
pi@raspberrypi:~ $ /home/pi/piKiss/piKiss.sh
```

Next time, please use the shortcut icon or  ```cd piKiss && ./piKiss.sh ```

The only thing I can do is put a warning and indicate the directory to be deleted before It proceed.

Fixed on https://github.com/jmcerrejon/PiKISS/commit/1c91582f344a7ab2b4053690f100f8c15b79ed5b.

closed 2020-08-27 by jmcerrejon