jmcerrejon/PiKISS #32: GPL2 copyright

sergiopoverony, 2020-09-27:
I see you wipe copyright from xash3d engine (for run hl1 and etc gold source mode) Why?

a1batross, 2020-09-27:
It doesn't look as GPL violation, it just never mentiones the tools used to run some games.

What's worse than that, seems most scripts for commercial game download data files from third-party servers, like archive.org. That's just a piracy.

a1batross, 2020-09-27:
Also what's interesting, engine reports that it's actually built for ARMv6 and script tells it's "optimized for RPi 4". 

With such old CPU instruction it will run flawlessly even on first RPi. :)

jmcerrejon, 2020-09-27:
Sorry @sergiopoverony . I'm just build scripts to make it works OOTB for non linux users. I'll fix the strings ASAP and add the copyright stuff. Maybe some of you guys could send me a Rpi4 optimized binaries (don't have time enough in my daily).

a1batross, 2020-09-27:
I would've build rpi4 binaries but don't have rpi unfortunately. There is enough arm boards in my life to get another one. :)

jmcerrejon, 2020-09-27:
Updated. Thank you all, guys. 😄

closed 2020-09-27 by jmcerrejon