jmcerrejon/PiKISS #36: Half-Life fresh install issues

rhester72, 2020-10-02:
1) client_armv6hf.so is not placed in valve/cl_dlls and thus cannot be found at startup (a workaround that seems like it would work regardless of whether the user has a pre-cached copy of the game content would be to create ~/games/half-life/cl_dlls and place it there, it seems to work just as well)

2) A pristine copy of the valve directory taken from a fresh Steam install does not result in a playable game - specifically, the menu items New Game and Training Room are ghosted and cannot be selected.

jmcerrejon, 2020-10-02:
Fixed. Now the ./half-life.sh check if some missing files and add it to your valve directory.

closed 2020-10-02 by jmcerrejon

rhester72, 2020-10-02:
Perfect, thank you!