jmcerrejon/PiKISS #37: zoom error on twister os

gji6frzujmmntzztztukjknvb, 2020-10-05:
Error loading needed lib: "libxcb-xfixes.so.0"
Error: loading needed libs in elf ./zoom

gji6frzujmmntzztztukjknvb, 2020-10-05:
sudo apt install libxcb-xfixes0
Paketlisten werden gelesen... Fertig
Abhängigkeitsbaum wird aufgebaut.       
Statusinformationen werden eingelesen.... Fertig
libxcb-xfixes0 ist schon die neueste Version (1.13.1-2).
0 aktualisiert, 0 neu installiert, 0 zu entfernen und 0 nicht aktualisiert.

Botspot, 2020-10-05:
I have noticed the same issue.
`libxcb-xfixes.so.0` does not get successfully copied to the `zoom` folder.
Only after I download it manually and copy it there does `zoom` work.

jmcerrejon, 2020-10-06:
Fixed on https://github.com/jmcerrejon/PiKISS/commit/4c0b6aae293eedc7506d82c8a0fc9bf0b943c3e6

closed 2020-10-06 by jmcerrejon

Botspot, 2020-10-06:
@jmcerrejon 
I don't see any changes you made that could have fixed this problem in the commit you mentioned.

jmcerrejon, 2020-10-06:
Fixed link!