jmcerrejon/PiKISS #38: no sound in zoom

pigithub0, 2020-10-08:
i'm getting no sound from zoom,downloaded from pikiss,also button "check speaker" not usable. Im using tv's speakers to output audio,for example on youtube sound works on my pi 4. It'just with me?pls answer

pigithub0, 2020-10-08:
for example,browser version of zoom works fine

jmcerrejon, 2020-10-08:
No one told me anything about that. This project just install the app and I use to test basic functionality and it's working on a clean latest RaspberryPi OS 32 bits. Maybe you can get some help on Pi Labs Discord's server or support visiting Zoom forums. Good luck! 🙂

closed 2020-10-08 by jmcerrejon

Botspot, 2020-10-24:
This is a pulseaudio problem. Zoom's audio input and output does not work at all without Pulseaudio.

Problem is, when you install pulseaudio on Raspbian, now other apps like Chromium won't play sound anymore!

In my [Pi-Apps app store](https://github.com/Botspot/pi-apps), I made this workaround: install pulseaudio when you install Zoom, but disable pulseaudio's autostart.  
When you launch my fixed version of Zoom, pulseaudio is executed beforehand. When you exit Zoom, pulseaudio is killed as well.

So far 100% of Raspbian users say that my approach with Pi-Apps works, while alternatives like Novaspirit's version and PiKISS's version don't work.

pigithub0, 2020-10-30:
thank you for your support,i almost given up with zoom with no sound,then i
can use full functionality of zoom.Second time,thank you!

Botspot, 2020-10-30:
> thank you for your support,i almost given up with zoom with no sound,then i can use full functionality of zoom.Second time,thank you!

Thank you as well!  
I am still shocked how both Novaspirit and @jmcerrejon could not noticed the audio problems - I mean surely at one of them would have tested the audio, right?

jmcerrejon, 2020-10-30:
Respect Botspot. Some comments can be turned against you. First advice.

Botspot, 2020-10-30:
> Respect Botspot. Some comments can be turned against you. First advice.

Thank you. Honestly I needed to hear that today. Comment edited.