jmcerrejon/PiKISS #41: Unalble to launch Half Life

jleboube, 2020-10-22:
Hi.  First, thank you for putting PiKiss together.  It is an amazing tool!

I recently tried to run Half-Life from PiKiss on my Raspberry Pi 4 (4GB RAM).  No matter what I try (Raspbian, Twister, etc), Half-Life won't launch.  I can install Quake and play it, but Half-Life is a no go. I do have Half-Life installed from Steam, however, through my testing, I tried with the Steam version installed and without.  The same results, which are, click on Games, then Half-Life.  The mouse just sits there and nothing happens.  Below is what I assume to be helpful information.  If you need me to get any log files or any other information, please let me know.

**Steps to reproduce:**
Log into Raspbian
Open terminal and run "sudo apt-get update"
Run "sudo apt-get upgrade -y"
Reboot
Paste "curl -sSL https://git.io/JfAPE | bash" into terminal and answer the questions during install
Reboot
Open menu, then go to Games, then click on Half-Life.  Nothing happens.

**gameinfo.txt:**
  GNU nano 3.2                      gameinfo.txt                                

// generated by Xash3D

basedir         "valve"
title           "Half-Life"
startmap                "c0a0"
trainmap                "t0a0"
version         1
type            "Single"
dllpath         "cl_dlls"
gamedll         "dlls/hl.dll"
gamedll_linux           "hl.so"
gamedll_osx             "dlls/hl.dylib"
icon            "game.ico"
sp_entity               "info_player_start"
mp_entity               "info_player_deathmatch"
secure          "1"
soundclip_dist  "1536"
hull0           ( 0 0 0 ) ( 0 0 0 )
hull1           ( -16 -16 -36 ) ( 16 16 36 )
hull2           ( -32 -32 -32 ) ( 32 32 32 )
hull3           ( -16 -16 -18 ) ( 16 16 18 )
max_edicts      900
max_tempents    500
max_beams               128
max_particles   4096

**xashds@.service:**
  GNU nano 3.2                    xashds@.service                               

[Unit]
Description=%I - Xash3D dedicated server
After=network.target

[Service]
User=xashds
Group=xashds
WorkingDirectory=/opt/xashds/srv/%I
EnvironmentFile=-/opt/xashds/srv/%I/env
ExecStart=/opt/xashds/bin/xashds $EXTRA_OPTS
Restart=always

[Install]
WantedBy=multi-user.target

jmcerrejon, 2020-10-22:
Try the next on Raspberry Pi OS 32 bits:

1. Install Half-life through _PiKISS_.
2. When asking for your online copy, type **n**.
3. Copy your _valve_ directory from _Steam_ inside _/home/pi/games/half-life_.
4. Now type in a terminal: ``` cd /home/pi/games/half-life && ./half-life.sh ```

jleboube, 2020-10-22:
Thank you for the prompt response. Can you kindly point me to where the valve and Steam directories are located?

jmcerrejon, 2020-10-22:
I mean open your _Steam folder_ (I don't know where it is in your operating system), search your _half-life/valve_ directory and copy it inside _/home/pi/games/half-life_

If you don't know where the games are in your Steam folder, you can search on the net.

jleboube, 2020-10-22:
Ok. So, here is one issue...I can’t install Steam on Raspbian, as it isn’t supported. I installed Twister OS, which has Steam installed. I will try to find the Half-Life folder and copy it over.

jleboube, 2020-10-22:
I followed your instructions and it still doesn’t work.

jmcerrejon, 2020-10-22:
Ok. I'll test it again with my Steam copy this week.

jleboube, 2020-10-22:
Please let me know if there are any logs or any other areas on my Pi I can share with you to help investigate/test.

I’ve tried this with a fresh copy of Raspbian and Twister OS on Raspberry Pi 4 with 4GB RAM. I like Twister and it has Steam pre-installed, so I’ve been using it. Im mentioning this in case Twister has a different log level we can turn on or any other tool to help with testing.

jmcerrejon, 2020-10-24:
It's working with no hassle. I've added some improvements. Check it out [on this video](https://youtu.be/Qq3rdfXfhY0).

closed 2020-10-24 by jmcerrejon