jmcerrejon/PiKISS #47: Autostart/Autoboot

actraiser30, 2020-11-09:
Hello,i want to install a game for example Dune2
and then make it everytime the OS boots to autostart/autoboot the game
is there such an option in your script?
if not can you add such an option to work with any of your games/servers etc ??
thanks

jmcerrejon, 2020-11-09:
About autostart/autoboot an app on Debian, you have a dozen tutorials about that on the internet. It's not difficult.

I've installed DuneLegacy in the latest RPi OS 32 bits, and it's working here with a USB mouse.

[image]

closed 2020-11-09 by jmcerrejon

actraiser30, 2020-11-09:
i tried the game in raspbian lite probably you tried it in the desktop image is that correct ?
if yes is there a way to work in the non desktop image?

also the starcraft is not shown in the games menu only i can see the expansion and diablo1 is not showing also only diablo 2 is there

jmcerrejon, 2020-11-09:
I always use Raspberry Pi OS 32 bits from here [here (direct link)](https://downloads.raspberrypi.org/raspios_armhf_latest) | [release notes](http://downloads.raspberrypi.org/raspios_armhf/release_notes.txt).

Maybe the issue is you don't have a desktop, try asking on Dune Legacy port [here](https://github.com/henricj/dunelegacy/issues).

Scripts are working here and I can see all the games. Try on the _Raspberry Pi OS_ I provided before and tell me if all issues are gone. That's the best advice I can give you now 🤷‍♂️ .

I'll add a note on README saying PiKISS is not compatible with Raspbian Lite.