jmcerrejon/PiKISS #50: PIKISS opens but doesn't install anything

Adam90a, 2020-12-05:
Hello!  I think my issue may have to do with using Raspberry Pi OS Lite with XFCE installed as my desktop environment.

I run the one line install, but it only opens PiKISS. I think I may be missing .desktop files in the right spot, but don't know the code to run PiKISS. 

Once I have it open, I try and install SuperMario64 and Halflife, both of which just send me back to the main PiKISS menu without the usual install pages and progress bars. Also no menu shortcuts created for them, but I may have to add .desktop files for those too, but as before, I don't know what codes to run them to put in the .desktop shortcuts. Also I am certain that the process failed.

Any additional info needed I can post, thanks for the help in advance!

jmcerrejon, 2020-12-05:
Probably is the change of desktop environment. Check the README file to run from Terminal.

Anyway. I need to test the recent changes on the OS. Keep you informed.

Adam90a, 2020-12-05:
Thank you! I am on the new update, I'll take a look at the readme now.

jmcerrejon, 2020-12-06:
All working as expected using default a clean latest Raspberry Pi OS Lite 32 bits. Desktop's shortcut with XFCE maybe have a different location, but anyway, you can cd into the directory of the game and run the executable.

closed 2020-12-06 by jmcerrejon