jmcerrejon/PiKISS #55: How to use this with other OS?

memidhun, 2020-12-14:
I am using Kali Linux in raspberry pi ..The pi kiss was successfully installed and it worked in terminal. I tried installing scrcpy using this .The program file was downloaded and installed but it's not opening whenever i run ./scrcpy it shows me "scrcpy no such file or directory" .I moved the entire folder to root folder but still it gave me the same error. I tried running it using the icon in the application menu but  it did't worked either.Is this somewhat related to the path because in kali linux home directory is in- /home/kali 
where as in rasbian-  /home/pi.

jmcerrejon, 2020-12-14:
Check the README file. PiKISS is only supported on Raspberry Pi OS 32 bits. Maybe it was installed, so search inside your partition where it is. Study the script for some tips and... good luck! 😊

closed 2020-12-14 by jmcerrejon

memidhun, 2020-12-14:
> Check the README file. PiKISS is only supported on Raspberry Pi OS 32 bits. Maybe it was installed, so search inside your partition where it is. Study the script for some tips and... good luck!

So it won't run on 64 bit ? How can i read the script?

jmcerrejon, 2020-12-14:
It doesn't support 64 bits atm. All the scripts are inside ~/piKiss/scripts

memidhun, 2020-12-14:
> It doesn't support 64 bits atm. All the scripts are inside ~/piKiss/scripts

oh is there any way i can use scrcpy in kali linux on pi? Its urgen! 
Thanks for helping me

jmcerrejon, 2020-12-14:
Sorry. I can't help you. I don't know anything about Kali. Good luck!