jmcerrejon/PiKISS #56: Possible older scrcpy is installed via PiKISS

Haledire, 2020-12-20:
I'm not sure what version of [scrcpy](https://github.com/Genymobile/scrcpy) that PiKISS installs, but it seems like some arguments from the latest version (1.16) don't work (specifically the --stay-awake argument).  I'm unsure if this is related to what version is installed and don't know where to look for this information.

I'm curious if there's anything particular about the version that PiKISS installs that would keep me from trying to install the latest version if need be to figure out this situation.

Haledire, 2021-01-05:
Just to clarify after doing some investigating - it seems PiKiss installs Scrcpy 1.13 (seems to be the same in TwisterOS)
I've had success in building 1.16 in Raspberry Pi OS via instructions from the Scrcpy github and from [linuxuprising](https://www.linuxuprising.com/2019/03/control-android-devices-from-your.html)

The reasons I needed 1.16 was a few small issues (one of which might be specific to my devices):
- `--stay-awake` (`-S`) was introduced in 1.14
- The ctrl+drag was not working for me at all in 1.13 but works fine in 1.16, though I noticed they'd reversed the ctrl and the mod keybinds at some point (everything that's done with ctrl in 1.13 is done with alt in 1.16)  Also it seems the left and right ctrl/alt buttons matter, so it could just be me trying to use the wrong key.
- I'd been having some issues with the commands to turn off the screen of the phone making it so the phone would not turn the screen back on when scrcpy was closed and the phone was unplugged.  This seems to become less of an issue with 1.16

Not a rush, just clarifying why I felt it was important to look into.

jmcerrejon, 2021-02-22:
Added new release 1.17. Closed with https://github.com/jmcerrejon/PiKISS/commit/c53b221fbfe501d2a105b23685bfa32baa11229c

closed 2021-02-22 by jmcerrejon