jmcerrejon/PiKISS #58: Half Life & Morrowind Problems

cokedolla, 2020-12-22:
Half Life>
New game, hazard course and load game are grayed out..it runs multiplayer though.

Morrowind>
Wont Launch..I get: Invalid desktop entry file: '/home/pi/.local/share/applications/openmw.desktop'

both work on my pc but cant get working on pi4 2gb..piOS

jmcerrejon, 2020-12-22:
HalfLife: Need to copy the valve dir onto the half-life directory

Morrowind: The binary files are in an inactive hoster (temporarily inactive due to high bandwidth demand). I need another one, but I can't find any free hoster to upload the files. I'll update this post thread when it has online again. Apologies.

cokedolla, 2021-01-03:
> HalfLife: Need to copy the valve dir onto the half-life directory
> 
confused by this instruction..so in pikiss dont link to the valve folder thats inside ..\pc games\Half-Life..link to ..\pc games\Half-Life itself?

jmcerrejon, 2021-01-03:
Check https://github.com/jmcerrejon/PiKISS/issues/59

closed 2021-01-03 by jmcerrejon