jmcerrejon/PiKISS #60: Caprice32 is very slow

notevenodd, 2020-12-28:
Caprice32 (Amstrad CPC emulator) installed from PiKiss is slow on a RaspberryPi 4. Emulation of simple games makes music all chopy and games are slow, both Xorg and cap32 processes use a lot of CPU. I had used Caprice previously on RetroPie setup on a Pi with no issue.

jmcerrejon, 2020-12-28:
If you have issues with the installation or it doesn't work, this is the right place. You can send your complaint to the [caprice32](https://github.com/ColinPitrat/caprice32) repo if you need a decent emulator or you could compile it by yourself. Make sure you tried to run:

- On the latest Raspberry Pi OS 32 bits.
- Full KMS support.
- No other app running in the background.
- Tried some settings to improve the speed.

Anyway, I'll try to run the emulator (I see it doesn't update for a long time) to check the speed with some games. What game did you try?

jmcerrejon, 2021-02-15:
Closed due inactivity

closed 2021-02-15 by jmcerrejon