jmcerrejon/PiKISS #62: Remove /home/pi/ in favor of %HOME%/ or ~/

RandamuMaki, 2020-12-31:
Please remove the /home/pi/ links in favor of XDG-standards so your scripts will use on Raspberry Pi machines who have removed the Pi account for security purposes or other.
Hard links should only be used for things like /usr/ and not /home/%USER%/ because %USER% may change per system.

jmcerrejon, 2020-12-31:
Maybe you see it in older scripts I couldn't review. I’ll check it out in a few days. Happy new 🎊

Pi-Mania, 2021-01-03:
This will fix a lot of issues i had using FenixOS also. (As there is no \home\pi user/folder) Older scripts i think. newer ones seem to work fine.

jmcerrejon, 2021-01-03:
I'm on it, but as I said in the README, PiKISS is only supported on RPi OS 32 bits, so use it at your own risk.

jmcerrejon, 2021-01-03:
Done!

closed 2021-01-03 by jmcerrejon

fenixlinuxos, 2021-01-03:
Thank you Jose. I was modifying pikiss and blocking the updates now it's not necessary.

fenixlinuxos, 2021-01-03:
Thank you Jose. I was modifying pikiss and blocking the updates, now it's not necessary.