jmcerrejon/PiKISS #66: Open Morrowind blank window

piotrfalba, 2021-01-02:
When I'm starting Open Morrowing I only had a window with borders - there is nothing else. Latest offcial Raspberry OS and Morrowind data bought from GOG.

jmcerrejon, 2021-01-03:
Working here with the latest Raspberry Pi OS 32 bits.

Make sure you run the game through the menu or using ./openmw.sh

Check the official repo if you have issues: https://gitlab.com/OpenMW/openmw/-/issues

Here is the list of files I have: [openmw-full-files.txt](https://github.com/jmcerrejon/PiKISS/files/5761427/openmw-full-files.txt)

Good luck!

Relintai, 2021-07-29:
Hi

Apparently the `libswscale.so.5` that comes with the current raspian is incompatible with the executable, and it's crashing while trying to play the bink video files.

I don't have an old raspian installation, so I wasn't able to test just adding the required .so files to the local ld library paths, so I'm not sure whether that will work or not, however I did managed to find a super simple workaround.

From: https://forum.openmw.org/viewtopic.php?t=6448

Open the `~/.config/openmw/openmw.cfg`.

find these lines:

```
fallback=Movies_Company_Logo,bethesda logo.bik
fallback=Movies_Morrowind_Logo,mw_logo.bik
fallback=Movies_New_Game,mw_intro.bik
fallback=Movies_Loading,mw_load.bik
fallback=Movies_Options_Menu,mw_menu.bik
```

Change the `.bik` filenames to something invalid, like:

```
fallback=Movies_Company_Logo,bethesda logoaaaaa.bik
fallback=Movies_Morrowind_Logo,mw_logoaaaaa.bik
fallback=Movies_New_Game,mw_introaaaaa.bik
fallback=Movies_Loading,mw_loadaaaaa.bik
fallback=Movies_Options_Menu,mw_menuaaaaa.bik
```

This way you won't have the intro videos, but at least the game will run.