jmcerrejon/PiKISS #67: Serious Sam The Second Encounter start issue

IronShell1207, 2021-01-03:
I cann't start SS TSE. 
I downloaded Serious Engine from "piKiss". Next, I downloaded the game itself from steam and copied it to the raspberry. Then I took the files from the "bin-fse" folder and copied them to the Bin folder in the game folder. And copied "SE1_10.gro" to the game folder. I go to the Bin folder and run the "ssam". The game starts loading files into memory, a splash screen appears, I press Enter to skip it. The game continues to download files and just crashes after a couple of seconds. 
I showed everything on the video https://youtu.be/2RT0GVX1Yo4

jmcerrejon, 2021-01-05:
Working here. As I mention in the script, follow the instructions to copy data files from https://github.com/ptitSeb/Serious-Engine

closed 2021-01-05 by jmcerrejon

Pi-Mania, 2021-01-06:
Ok i think i have a FIX/Workaround for the exit before game starts problem. Just finalising testing but looking good. Jose, the pi-kiss script creates incorrect shortcut names. (for sam2 at least)  Possibly some issue if i say i have the sam 2 game files, it doesnt seem to download the ssam-bin1.05 files?  Thanks