jmcerrejon/PiKISS #68: OpenMW not downloading as Readonly missing from Script File

Pi-Mania, 2021-01-05:
readonly missing from beginning of BINARY_URL line so Openmw fails to start downloading. 😁

jmcerrejon, 2021-01-05:
BINARY_URL can't be read only because is overwrite if you have the full version. Make sure the file is online on your res/magic-air-copy-pikiss.txt

Pi-Mania, 2021-01-05:
Its the openmw files that fail to download and on Twister also. My morrowind files copy to games folder just fine.  Maybe i explain badly, test and you will see what i mean. Thanks

jmcerrejon, 2021-01-05:
I run the script, then I press n to the Magic Air Copy message, and then start to download the binary files. Anyway, read the BINARY_URL and copy the URL, then paste it in the browser (or wget it). The file is online right now.

closed 2021-01-08 by jmcerrejon