jmcerrejon/PiKISS #74: AVP no sound

retro24, 2021-01-07:
Hello, i have installed avp from your script in my RPI 4 it runs fine but i dont have sound in the game?

do you know whats wrong ??

retro24, 2021-01-07:
In Raspberry pi 3 the sound works in Raspberry pi 4 does not work,so you must have build the executable in an Rpi3 or not?

retro24, 2021-01-08:
update

installed pulseaudio and enable it now i have sound in the game from the headphones only need to figure out how to get it to the speakers

jmcerrejon, 2021-01-08:
So the problem is on your side. I’ll close the issue. Good luck!

symbios24, 2021-01-08:
Its not in my side the problem is general i suggest you try and see for your self and fix it, if you can,its a general rpi4 problem

jmcerrejon, 2021-01-08:
Ok. I’ll try it and tell you.

jmcerrejon, 2021-01-10:
Tested. It works. Sound effects are the only ones that work (I never tested the music). Good luck!

closed 2021-01-10 by jmcerrejon