jmcerrejon/PiKISS #75: PiKiss not starting up

Surfxman, 2021-01-07:
Hello

I am hoping you can help me please.

First i am novice, but happy to learn. I got myself a 8 GB RPI, with the Argon M2 case. I have installed Twister OS and have updated to version 1.9.6 and i have a wired connection.

When i double click on the PiKiss icon, a terminal window pops up with the following message:

ping: socket: Operation not permitted

Internet connection required. Causes:

A. Check your network
B. Weak WiFi signal?
C. Try no check internet connection parameter (-ni): cd ~/piKiss && ./piKiss.sh -ni

Press [Enter] to exit...

Well, my network is fine and i am not using wifi (wired connection).
So, first i didnt understand to do with C. I then opened up terminal, cut and pasted the script and pressed enter.
PiKiss now opens up in terminal and is fully useable. I even checked the internet option on PiKiss and it returned DL 200+ .

How can i fix this please ?

Thank you

jmcerrejon, 2021-01-08:
Did you update your distro? Search a howto on the internet and try again. Search on close issues, I remember a user asked the same and the issue was on his side. I’ll close this issue in a couple of days because is not a problem with PiKISS.

Surfxman, 2021-01-09:
> Did you update your distro? Search a howto on the internet and try again. Search on close issues, I remember a user asked the same and the issue was on his side. I’ll close this issue in a couple of days because is not a problem with PiKISS.

Thank you for responding.

Whilst i am not 100% sure what you mean, I believe i have updated everything. On first install of Twister OS, i upgraded to the latest version of Twister OS (v 1.9.6) and then within a terminal:

(1) sudo apt update
(2) sudo apt-get upgrade
(3) sudo apt-get dist-upgrade

This then got me to the current problem i have.

I even tried to execute the file within file explorer still the same issue.
Only when i use "cd ~/piKiss && ./piKiss.sh -ni" in terminal, does PiKiss work.
My internet is working and so the script has to be bypassed (-ni) for it to work ? 

As i mentioned, i am a novice and so don't really know how to proceed further and am just guessing.
My next step is to uninstall and re-install.

Regards

jmcerrejon, 2021-01-09:
I remember a user asked the same on the Discord server. I encourage you to join us there and ask on #pikiss-support channel. Good luck!

closed 2021-01-10 by jmcerrejon