jmcerrejon/PiKISS #78: half life something is wrong aborting... on installation

333337lee, 2021-01-16:
Hi Jose,

I have tried 4 different installs or Raspberry Pi OS 32bit. One was a clean install from Raspberry Pi imager.

all come up with the message in the title. I have had it working in the past. seems to be affecting serious sam and quake 2 as well.

Duke Nukem worked fine.

Regards Lee
Leepspvideo

jmcerrejon, 2021-01-16:
Could you attach a screenshot with the terminal and the message?. I’ve installed in the latest RPi OS answering N for magic air copy, then I’ve copied my /valve directory and running /half-life.sh, the game was working as expected. Same with Quake 2. Nothing has changed on those scripts and the files are online. You could try to download the binaries (open the script and copy the content of BINARY_URL). Then wget to download the file, uncompress and copy inside the game’s directory your data files. There is no mystery 😅

333337lee, 2021-01-17:
Thanks, is the Y option no longer supported?

[image]

jmcerrejon, 2021-01-17:
Due to legal stuff, I can't provide the valve directory with data games, so you need to add it. You have two options: 1. Do it manually once the binaries with the engine are installed. 2. If you have the directory compressed on the cloud or in your file system, set the URL/path in a file inside /res/magic-air-copy-pikiss.txt directory. For more info, check the README file in this repository.

333337lee, 2021-01-17:
Thank you, I hadn’t realised it had changed.

regards Lee.

jmcerrejon, 2021-01-27:
np

closed 2021-01-27 by jmcerrejon