jmcerrejon/PiKISS #104: Starting PiKiss

DickBurke, 2021-05-28:
When I try to start PiKiss I get the following error message:
Failed to execute child process "xterm" (no such file or directory)
I am trying to set up a Raspberry Pi 4 with zoom for my grandson, age 7, 1500 miles away. Please help!
Thank you, Dick Burkhartzmeyer

jmcerrejon, 2021-05-28:
Please use PiKISS on Raspberry Pi OS 32 bits.

DickBurke, 2021-05-28:
I'm sorry, I don't understand your comment. I used the following command:
curl -sSL https://git.io/JfAPE | bash
to install PiKiss. When I tried to run it the messages I sent you before is what I got. Am I doing something wrong?
Dick Burkhartzmeyer

jmcerrejon, 2021-05-29:
What operating systems are you using?

DickBurke, 2021-05-29:
I am using the latest and greatest update to the raspberry pi OS. My machine is not near me right now so I cannot give you the version number from the computer.
Dick Burkhartzmeyer

jmcerrejon, 2021-05-29:
It should be installed xterm, but because an unreasonable true, is not installed on your system 🤷‍♂️ 

Possible solution: Install xterm by yourself.

`sudo apt get install -y xterm`

closed 2021-06-15 by jmcerrejon