jmcerrejon/PiKISS #105: Raspbian Buster 64 bits scrcpy error

pm-alonso, 2021-06-15:
Hola, cuando intento usar el scrcpy me da el siguiente error:

./scrcpy: error while loading shared libraries: libavformat.so.58: cannot open shared object file: No such file or directory

alguien sabe como resolver esto ?

Gracias.
Saludos

pm-alonso, 2021-06-15:
Oh I see everybody writes in english in this town... just in case is needed I will rewrite my question.

I am experienciong this error trying to run scrcpy installed with PiKISS script...

./scrcpy: error while loading shared libraries: libavformat.so.58: cannot open shared object file: No such file or directory

thanks for your help.
Regards,

jmcerrejon, 2021-06-15:
Apologies. PiKISS is not yet 64 bits capable.

pm-alonso, 2021-06-16:
OK Jose, thank you for your effort and sharing such a great tool anyway...

I would like to know how to compile this scrcpy to my raspberry pi 4 running with buster 64bits....
Is there any change you can help me to accomplish this ? I do not know much of programming, just a bit of traditional cmd batch (M$) and bash (GNU/Linux).

I tied to follow build instructions of scrcpy project in github directly but this two steps doesn't seems to work for me:

pi@mypi64:~/scrcpy $ sudo apt install openjdk-8-jdk
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package openjdk-8-jdk

pi@mypi64:~/scrcpy $ make -f Makefile.CrossWindows
make: Makefile.CrossWindows: No such file or directory
make: *** No rule to make target 'Makefile.CrossWindows'.  Stop.

thanks.
regards,

Pablo

jmcerrejon, 2021-07-17:
Check inside my script a function called compile and copy/paste each command onto your Terminal. Good luck!

closed 2021-07-17 by jmcerrejon