jmcerrejon/PiKISS #111: DiabloII

max0r4axor, 2021-08-21:
The desktop entry that is created points to the wrong directory for the diablo2.sh and diabloII.png thumbnail.

jmcerrejon, 2021-08-21:
On RPi OS?. It's the only OS supported

max0r4axor, 2021-08-21:
Yes on RPi OS. The files are created in the folder /home/pi/games/diablo2/ but the desktop entry points to /home/pi/diablo2/

max0r4axor, 2021-08-21:
I see the issue its in the piKiss/scripts/games/diablo2.sh line 50 and 51
Exec=${PWD}/diablo2/diablo2.sh
Icon=${PWD}/diablo2/diabloII.png
it should be 
Exec=${PWD}/games/diablo2/diablo2.sh
Icon=${PWD}/games/diablo2/diabloII.png

jmcerrejon, 2021-08-21:
No worry. I got the bug, but I want to refactor the script a bit. Thanks for that.

jmcerrejon, 2021-08-28:
Fixed https://github.com/jmcerrejon/PiKISS/commit/8783577c3fa5ab0ddee84d48de8da4b29bfa25c2

closed 2021-08-28 by jmcerrejon