jmcerrejon/PiKISS #112: Please update DevilutionX to 1.2.1

AJenbo, 2021-09-15:
This adds Hellfire support as well as other improvements and fixes.
See https://www.reddit.com/r/Devilution/comments/pm1mqk/devilutionx_update_hellfire_fail/ for more details.

Where:
https://github.com/jmcerrejon/PiKISS/blob/28e6558d59a59991f9d58dcd27d4194d7727e1b6/scripts/games/diablo.sh#L17

New URL:
https://github.com/diasurgical/devilutionX/releases/download/1.2.1/devilutionx-linux-armhf.tar.xz

jmcerrejon, 2021-09-18:
https://github.com/jmcerrejon/PiKISS/commit/d8c0eae12ac3470c06c19ad613583eb923b942b1

closed 2021-09-18 by jmcerrejon

AJenbo, 2021-11-07:
1.3.0 is now available which lets you Switch between Hellfire and Diablo ingame as well as a bunch of other updates: https://github.com/diasurgical/devilutionX/releases/tag/1.3.0