jmcerrejon/PiKISS #115: installation script is exiting with non-zero exit code

Botspot, 2021-09-24:
Pi-Apps users routinely submit error reports because PiKISS is reporting itself as failing.

One example error log we've received:
```
OS: Raspbian GNU/Linux 10 (buster)
OS architecture: 32-bit
Pi-Apps local commit ID: 15c6037244281c6aa8d17b2aeccd9413c134075a
Kernel: armv7l 5.10.60-v7l+
Device model: Raspberry Pi 4 Model B Rev 1.4
Raspberry Pi OS image version: 2021-05-07
Language: en_US.UTF-8

BEGINNING OF LOG FILE:
-----------------------

Installing piKiss...
HJJHJJ...
From https://github.com/jmcerrejon/PiKISS
   a6f9bdf..d37b03a  master     -> origin/master
   a6f9bdf..d37b03a  develop    -> origin/develop

 New version available!

 \u00b7 Installing updates...

Fetching origin
HEAD is now at d37b03a \U0001f4d6 DOC: Add Game Shadow Warrior. Diablo now support Hellfilre.
From https://github.com/jmcerrejon/PiKISS
 * branch            master     -> FETCH_HEAD
Already up to date.

PiKISS is up to date!. 

You need to run the program again.

Failed to install piKiss!

Failed to install piKiss with install-32 script!
\u25b2 Need help? Copy the ENTIRE terminal output or take a screenshot.
Please ask on Github: https://github.com/Botspot/pi-apps/issues/new/choose
Or on Discord: https://discord.gg/RXSTvaUvuu
```
Pi-Apps [uses](https://github.com/Botspot/pi-apps/blob/master/apps/piKiss/install-32#L10) this command to install PiKISS:
```
curl -sSL https://git.io/JfAPE | bash || error 'Failed to install piKiss!'
```

jmcerrejon, 2021-09-24:
I think the issue is that PiKISS is already installed and the user tries to install it again over a copy. I’ll check it out, but you could check if it was installed previously.

Botspot, 2021-09-24:
> I think the issue is that PiKISS is already installed and the user tries to install it again over a copy. I’ll check it out, but you could check if it was installed previously.

I just tried it and if pikiss is already installed, that curl command will just launch pikiss.
Then I uninstalled pikiss, and tried running this command:
```
curl -sSL https://git.io/JfAPE | bash ; echo "exit code: $?"
```
And sure enough, the exit code came back as 1.

jmcerrejon, 2021-09-24:
Test now and tell me. I hope the commit https://github.com/jmcerrejon/PiKISS/commit/940f2ea67ce51b0400d2687fd5f9d85acb0d88b8 help us 🤞

Botspot, 2021-09-24:
> Test now and tell me. I hope the commit [940f2ea](https://github.com/jmcerrejon/PiKISS/commit/940f2ea67ce51b0400d2687fd5f9d85acb0d88b8) help us 🤞

The exit code is now zero.
```
...
remote: Enumerating objects: 27, done.
remote: Counting objects: 100% (27/27), done.
remote: Compressing objects: 100% (4/4), done.
remote: Total 18 (delta 14), reused 18 (delta 14), pack-reused 0
Unpacking objects: 100% (18/18), done.
From https://github.com/jmcerrejon/PiKISS
   d37b03a..d9af4da  master     -> origin/master
   d37b03a..d9af4da  develop    -> origin/develop

 New version available!

 \u00b7 Installing updates...

Fetching origin
HEAD is now at d9af4da \U0001f4d6 DOC: Minor bug fixes on installation and Zandronum source code path
From https://github.com/jmcerrejon/PiKISS
 * branch            master     -> FETCH_HEAD
Already up to date.

PiKISS is up to date!. 

You need to run the program again.

exit code: 0
```

closed 2021-09-24 by Botspot