jmcerrejon/PiKISS #116: Problem with Diablo2

jackygoule, 2021-09-25:
Hello, i have a problem when D2 start ,after the Blizzard's logo .
On the start page the error :erreur diablo 2 halt location:,line#260 Expression:Unrecoverable internal error 410f8f3
[image]
4

Thanks for your incredible job .

jackygoule, 2021-09-25:
Hello i found solution.
On the game in windows after basic install, update  the game for battle.net  and recover the file :   Patch_D2.mpq
After replace your file :  Patch_D2.mpq     with the new in your install on pi.

A small problem stay to me when i finish my game my desktop have a wrong resolution and i reboot for rescue my config.

jmcerrejon, 2021-09-25:
Ok. So that's depends on what's the data files you provide. I never got that problem. 🤷‍♂️ 

I don't know If It's a problem I can solve, but thanks for the info. Very valuable.

closed 2021-09-28 by jmcerrejon