jmcerrejon/PiKISS #117: box86 install problem

actraiser30, 2021-09-25:
Hello i have a Raspberry pi4 with latest raspbian buster desktop

im trying to compile box86 from source and its not possible i always get a message saying: 

already up to date

your box is already updated skipping

i have not installed box86 in anyway eg: from binary how is up to date ?

must be a bug can you check it please

thanks

jmcerrejon, 2021-09-25:
Yes, you have. Check If you have the directory _/home/pi/box86_. 😉

actraiser30, 2021-09-25:
that command  returns nothing
and as i said i have not install it

jmcerrejon, 2021-09-25:
That command is used for check if it's installed. If return the path, then you have install it and show that message. Very simple, so I can't do anything more, apologies.

closed 2021-09-25 by jmcerrejon

actraiser30, 2021-09-25:
you are weird anyway thanks

jmcerrejon, 2021-09-25:
I've tested on my Pi and you are right, @actraiser30.

The point is that line should be working...Mm, I need to test it deeply.

My apologies, but I will not allow insults and I have deleted your previous message offending me. I will try to fix it when I have 
 enough time.

actraiser30, 2021-09-25:
Ok thank you and sorry

actraiser30, 2021-09-25:
If its possible to update also the pre- compiled version its almost a year
old
Thanks

jmcerrejon, 2021-09-26:
Fixed b27001663c5d3fcf986a859d119a4742cb53d9a9

commit 32b1af3b (2021-09-25): 📖 DOC: Fix #117. Refactor call to make_install_compiled_app