jmcerrejon/PiKISS #127: Vulkan installation fails in Bulsseye

Qengineering, 2021-11-15:
The Vulkan installation always worked perfect in Buster.
However, now with Bullseye, it errors. 
A few libraries can't be installed and Ninja didn't start to compile.

- Raspberry Pi 4
- 2 GB RAM
- 64 bit Bullseye OS

jmcerrejon, 2021-11-15:
Sorry, but at the moment, I don't give support for 64 bits OS. My apologies 😢

Qengineering, 2021-11-15:
Sorry @jmcerrejon,

The same applies for the 32 bits OS version of Bullseye.

The python_mako can't be loaded due to the dropping of python2 in Bullseye. 
python3_mako could be your subsitute. 
However, ninja fails to run. 

- Raspberry Pi 4
- 2 GB RAM
- 32 bit Bullseye OS

jmcerrejon, 2021-11-16:
Fixed afd51d95d537555012336bcc1f208dfdf1700c50

closed 2021-11-16 by jmcerrejon

Qengineering, 2021-11-17:
A friendly request.
I know you mentioned not supporting the 64-bit OS of the Raspberry Pi.
But could you make an exception in the case of the Vulkan installation?

Let me explain. Someone mentioned to me that my tutorial on how to install Vulkan on a Raspberry Pi is "Best Practice" at the [Vulkan site](https://vulkan.org/learn). OK, nice.

The first thing I mention on my site is, explaining PiKISS as an excellent installation tool for Vulkan. 
However, now knowing it's only for 32-bit, I understand the mails with the ERROR_EXTENSION_NOT_PRESENT error when using the Vulkan engine. It's generated when you follow the 32-bit installation on a 64-bit machine. 

There are three options.

1. I make a note in the text explaining PiKISS only works on a 32-bit OS
2. The PiKISS software detects a 64-bit OS and stops installing Vulkan with an exception or something.
3. The PiKISS software detects a 64-bit OS and uses only `CFLAGS="-mcpu=cortex-a72" CXXFLAGS="-mcpu=cortex-a72"`

See the guide [here](https://qengineering.eu/install-vulkan-on-raspberry-pi.html).

jmcerrejon, 2021-11-17:
Let me study that, right?

Qengineering, 2021-11-17:
Fine with me.