jmcerrejon/PiKISS #130: StarCraft launch error.

Rak1ta, 2021-12-04:
The game will play normally if Wine is not present on the Pi at the time of installation. 
But I have wine from the Pi-Apps store. And when I try to install StarCraft after that, an error occurs:
```
pi@raspberrypi:~ $ /home/pi/games/starcraft/starcraft.sh
Box86 with Dynarec v0.2.5 f79e16f4 built on Nov 22 2021 22:36:18
Box86 with Dynarec v0.2.5 f79e16f4 built on Nov 22 2021 22:36:18
Box86 with Dynarec v0.2.5 f79e16f4 built on Nov 22 2021 22:36:18
0024:err:module:__wine_process_init L"C:\\windows\\system32\\libscr_sa_arm.exe.so" not found
pi@raspberrypi:~ $ /home/pi/games/starcraft/starcraft.sh
Box86 with Dynarec v0.2.5 f79e16f4 built on Nov 22 2021 22:36:18
Box86 with Dynarec v0.2.5 f79e16f4 built on Nov 22 2021 22:36:18
Box86 with Dynarec v0.2.5 f79e16f4 built on Nov 22 2021 22:36:18
0024:err:module:__wine_process_init L"C:\\windows\\system32\\libscr_sa_arm.exe.so" not supported on this system
```
Why is this happening?

jmcerrejon, 2021-12-04:
When you install it, it tells you the reason: 
_This game uses Wine from the repo_

You have a conflict with wine versions installed: One is used for x86 binaries and the other for arm compiled apps. You can only have installed one of them. 

Question: Are you using Twister OS?

Rak1ta, 2021-12-04:
@jmcerrejon Raspberry Pi OS 32bit Debian10

jmcerrejon, 2021-12-04:
Thanks for the feedback.

I can't do anything with that, except showing a warning message to the user If it's using wine for x86.

I let this issue open until I include this message.

Rak1ta, 2022-07-02:
@jmcerrejon Is Diablo 2 the same? The game requires Wine.