jmcerrejon/PiKISS #131: Installing PSX (Duckstation)

Rak1ta, 2021-12-06:
**BIOS_URL="http://185.232.52.56 -** does the site work? my installation hangs at that point. there is no response from the site.
[psx.sh](https://github.com/jmcerrejon/PiKISS/blob/master/scripts/emus/psx.sh)

jmcerrejon, 2021-12-07:
Fixed 1871d65407e07781247eecd019accd4d35df6ab0

closed 2021-12-07 by jmcerrejon

Rak1ta, 2021-12-07:
@jmcerrejon installation completed but launch was unsuccessful
```
pi@raspberrypi:~ $ /home/pi/games/duckstation/duckstation-qt
/home/pi/games/duckstation/duckstation-qt: /lib/arm-linux-gnueabihf/libpthread.so.0: version `GLIBC_2.30' not found (required by /home/pi/games/duckstation/duckstation-qt)
/home/pi/games/duckstation/duckstation-qt: /lib/arm-linux-gnueabihf/libm.so.6: version `GLIBC_2.29' not found (required by /home/pi/games/duckstation/duckstation-qt)

```
Doesn't work on Buster?

jmcerrejon, 2021-12-07:
I'm compiling in the new Bullseye, so you confirm that It doesn't works on Buster... I don't want to come back to Buster, but I'll check it out If I can do anything