jmcerrejon/PiKISS #135: RPI PLAY - doesn't load to black screen - Ubuntu 20.04

SilverGoldCharzard, 2021-12-23:
I don't have a black screen
How can I connect to rpiplay

jmcerrejon, 2021-12-23:
Maybe any dependencies are not satisfied, because you are using Ubuntu, and I compiled RPiPlay and make the script for Raspberry Pi OS Buster 32 bits (the only OS PiKISS is supported).

So I don't think this could be working on Ubuntu. You can check the script and try to compile by yourself on it. 

Good luck!

SilverGoldCharzard, 2021-12-23:
> Maybe any dependencies are not satisfied, because you are using Ubuntu, and I compiled RPiPlay and make the script for Raspberry Pi OS Buster 32 bits (the only OS PiKISS is supported).
>
> So I don't [...]

Also doesn't work on Raspberry Pi OS Bullseye

SilverGoldCharzard, 2021-12-23:
> Maybe any dependencies are not satisfied, because you are using Ubuntu, and I compiled RPiPlay and make the script for Raspberry Pi OS Buster 32 bits (the only OS PiKISS is supported).
>
> So I don't [...]

Where is the build script for it

jmcerrejon, 2021-12-23:
piKiss/scrips/others/rpiplay.sh is the path

SilverGoldCharzard, 2021-12-23:
> ~/piKiss/scripts/others/rpiplay.sh is the path

Thats litterry the script that is ran when selecting rpiplay in Pi Kiss

jmcerrejon, 2021-12-23:
Yeah. You can read and see how to compile on Ubuntu or whatever you want. Maybe I'll try to compile today, If you can wait for the Bullseye version.

SilverGoldCharzard, 2021-12-23:
I couldn't figure out how to get the renderer to work

jmcerrejon, 2021-12-23:
OK. It's not as easy as I thought. RPi OS Bullseye removes the OpenMAX renderer and now I need to use gstreamer. Could you test the next binary I compiled?:

Install the next two new dependencies: `sudo apt install -y libgstreamer1.0-0 libgstreamer-plugins-base1.0-0`

Download, uncompress and run the next on your _RPi OS 32 bits Bulsseye_: https://misapuntesde.com/rpi_share/rpiplay-testing.tar.tz

Keep me informed.

jmcerrejon, 2021-12-23:
Related https://github.com/FD-/RPiPlay/issues/308

SilverGoldCharzard, 2021-12-23:
> Related https://github.com/FD-/RPiPlay/issues/308

yes

SilverGoldCharzard, 2021-12-23:
> OK. It's not as easy as I thought. RPi OS Bullseye removes the OpenMAX renderer and now I need to use gstreamer. Could you test the next binary I compiled?:
>
> Install the next two new dependencies: [...]

can't seem to get the file to run

jmcerrejon, 2021-12-23:
> 
> can't seem to get the file to run

Uncompress the file, run a Terminal and show me the output when you run `./rpiplay` inside the directory where you have the file (probably _~/Downloads_).

SilverGoldCharzard, 2021-12-23:
> > can't seem to get the file to run
> 
> Uncompress the file, run a Terminal and show me the output when you run `./rpiplay` inside the directory where you have the file (probably _~/Downloads_).

bash: ./rpiplay: No such file or directory
pi@raspberrypi:~/Downloads $

jmcerrejon, 2021-12-23:
Sorry, I didn't know you don't have the knowledge to uncompress a file. I'll try to test in a clean OS and tell you in a couple of days.

jmcerrejon, 2021-12-24:
Tested the file I sent [here](https://github.com/jmcerrejon/PiKISS/issues/135#issuecomment-1000485233) and on a new installation of _RPi OS 32 bits Bullseye_ and...it's working!. Installing the deps I said. I'll try to upload the changes today/tomorrow.

[image]

jmcerrejon, 2021-12-25:
Fixed d0268f33ef5a9dffd049dd83fd296091b9c97c4b

closed 2021-12-25 by jmcerrejon