jmcerrejon/PiKISS #141: Pikiss and Vulkan Driver integration on Retropie v.4.7.20 with pi4, is it possible to do ?

Giandj, 2022-02-08:
Hi I have Raspberry PI4 8gb where I have installed latest version of Retropie v.4.7.20, and Pixeldesktop installed, so I have installed PiKiss,
and next I have installed emulator PPSSPP and Vulkan drivers support (by Pikiss from the GUI), but when I open my retroarch v.1.10.0
I cannot see any Vulkan integration ? Is it correct ?, on Retroarch (launched by Emulationstation) there is no choice to change GL dirver to Vulkan ? Infact if I launch PPSSPP emulator from Emulationstation menu,  I cannot see "Vulkan" but only "OpenGL", instead if I use PPSSPP installed by PiKIss (from Pixeldesktop) I can correctly modify video driver from OpenGL to Vulkan , and on pixeldesktop PPSSPP emulator works very fine with Vulkan driver (seems I have 2 different installation of PPSSPP.
Why in Retroarch I cannot see Vulkan support ? 
There is a method to integrate the already installed Vulkan driver also on retroarch previosly installed ?
Thanks

jmcerrejon, 2022-02-08:
Hi. Did you try the next?

`sudo apt install -y mesa-vulkan-drivers libvulkan-dev libvulkan1 vulkan-tools`

Giandj, 2022-02-09:
Yes i have tried but nothing happens..
On retropie , retroarch cannot see vulkan...
Maybe i need to modify retroarch.sh.
But what parameter ? I need to change on retroarch.sh and reinstall retroarch from source ?

jmcerrejon, 2022-02-09:
Mm, Are you using Raspberry Pi OS 32 bits?. That's the only way I tested the driver. Beyond that, sorry but I can't help you 😞

Giandj, 2022-02-18:
I use Retropie (raspian buster) v.4.7.20
With pixeldesktop installed
But is it normal that if i had installed retroarch, emulationstation and so on, in /home/pi/Retropie/ directory 
When i have also installed retroarch on pikiss now i have 2 retroarch ?
There is a method to merge retroarch installed by pikiss with pre-existing retroarch in /home/pi/Retropie/ ?
Otherwise with 2 retroarch installed they make a conflict with joystick and configuration of the games...

jmcerrejon, 2022-03-05:
Nothing to do here. I close the issue.

closed 2022-03-05 by jmcerrejon