jmcerrejon/PiKISS #150: OBS launch failed

XavierColombel, 2022-05-17:
I'm on a Pi 4 with Bullseye 32bits installed.

[image]

I've installed OBS but I failed to launch it. Here is the error I got:

[image]

Don't focus on those 2 lines as I took last screenshot from a VNC, but the whole error is the same:

[image]

jmcerrejon, 2022-05-19:
Did you try? 

```sh
MESA_GL_VERSION_OVERRIDE=3.3 obs
```

jmcerrejon, 2022-05-19:
Tested with the latest command and It works. Now you have it on 64 bits, too 😉

closed 2022-05-19 by jmcerrejon

XavierColombel, 2022-05-20:
I confirm, on 64 bits, it's on fire ! 🔥 Thanks !

XavierColombel, 2022-05-20:
@jmcerrejon did you test on a 32 bits distribution ? It's not working for me, same error about the `segmentation fault`, even with the `MESA_GL_VERSION_OVERRIDE=3.3 obs` command.