jmcerrejon/PiKISS #154: QUAKE 1 won't start

Rak1ta, 2022-06-23:
```
Press [Enter] to install the game...
Installing Quake, please wait...
Installing binary files...
Downloading...
quakespasm-0.94.3-r 100%[===================>] 764,89K  3,29MB/s    за 0,2s    
Extracting...
Generating icon...
Downloading...
quake106.zip        100%[===================>]   8,67M  5,10MB/s    за 1,7s    
Extracting...
Done!. You can play typing /home/pi/games/quake/run.sh or opening the Menu > Games > Quake.

File does not exist.
· Something is wrong.
· Try to install again.

Press [Enter] to go back to the menu...

```
`QUAKE ERROR: W_LoadWadFile: couldn't load gfx.wad` - The demo was supposed to load itself?

i use: RaspiOS 32bit Debian10

Rak1ta, 2022-06-27:
`./quakespasm: ./quakespasm: unable to run binary file`

jmcerrejon, 2022-07-21:
It works on latest Bullseye 32-bit. I have to fix the shareware game data version.

[image]

Rak1ta, 2022-07-21:
@jmcerrejon it works for me

closed 2022-07-30 by jmcerrejon