jmcerrejon/PiKISS #155: DOSBox-X won't start

Rak1ta, 2022-06-23:
Program not working after [update](https://github.com/jmcerrejon/PiKISS/commit/19bde65b383fd187e34967c4d003da7ef171ced7).
Requires at startup:
```
Press [ENTER] to run...
./dosbox-x: error while loading shared libraries: libslirp.so.0: cannot open shared object file: No such file or directory
```
Such a package is missing from Buster
https://packages.debian.org/bullseye/libslirp-dev

i use: RaspiOS 32bit Debian10

jmcerrejon, 2022-06-25:
Fixed with a95884cddca3030935bfcef46eace7d0ca09d2bc

I'm not giving support to Debian 10, sorry.

closed 2022-06-25 by jmcerrejon

Rak1ta, 2022-06-25:
@jmcerrejon I know you only have Bullseye :(

Rak1ta, 2022-06-27:
@jmcerrejon  libfluidsynth2 - can be added to dependencies, requires `libfluidsynth.so.2`