jmcerrejon/PiKISS #159: Duckstation invalid archive

Rak1ta, 2022-06-29:
`readonly BIOS_URL="https://dl.hexrom.com/rom/psx-bios-SCPH1001-hexrom_com.zip"`

This is the PCSX2 installer: 
[image]

symbios24, 2022-07-15:
Hello there can you update the Duckstation binary to the latest version please?
Thank you

symbios24, 2022-07-19:
There isn't a 32bit latest version of duckstation?

Im not using 64bit os and not planning to use,as its not behaving/working correct and has many incompatibilities,if you want you can rollback to the previous 32bit version of duckstation
Thank you

jmcerrejon, 2022-07-19:
I'm not planning to compile the latest version for Buster/Bullseye 32 bits. The current for 32 bits is the old one I already compiled... I'll ask to my patrons and they will decide on its retro compatibility.

symbios24, 2022-07-19:
I mean to put back the previous version you already had before i did ask to compile any other

jmcerrejon, 2022-07-19:
It's the same older version. I only compiled the 64 bits available at Gihub 🤷‍♂️

jmcerrejon, 2022-07-21:
Fix Buster shortcut. 

fb8b6c80edff55b451ef2782423b04a0e6fb61b7

I close the issue finally. I'll think about upgrading/compiling for 32 bits. Thanks.

closed 2022-07-21 by jmcerrejon