jmcerrejon/PiKISS #165: Vulkan driver - crash system - rasp pi4 64bit

douglasdock, 2022-09-09:
After installing the latest vulkan driver that is available, the system crashes after a few reboots. I realized it was after I tried to use the vulkan drive to play games.

bagong, 2022-10-08:
I can confirm that a 64-bit system with all bullseye updates cannot boot anymore after running the script using the "stable" 22.1 branch

jmcerrejon, 2022-10-09:
unfortunately, it's not a problem with my script. Maybe I should choose a more stable branch. What do you think, guys?

darksaviorx, 2022-10-13:
I see that the active branches are 21.3, 22.1, 22.2 and main. I'm not counting the staging branches. 
Maybe offer the other branches as well to satisfy everyone? The 22.2 branch got a lot of fixes today and I would love to test that one out.

jmcerrejon, 2022-10-13:
> I see that the active branches are 21.3, 22.1, 22.2 and main. I'm not counting the staging branches.
> Maybe offer the other branches as well to satisfy everyone? The 22.2 branch got a lot of fixes today and I would love to test that one out.

I updated the script and now it takes into account branch 22.2. You guys are free to test and tell me If it works (we all hope so). Thanks.

jmcerrejon, 2022-10-16:
It seems we continue with the Desktop crash issues... very lost about it. The only provisional fix is run again inside ~/mesa_driver sudo make install

douglasdock, 2022-10-17:
My friends, if I install the driver and crash or no longer enter the operating system, is there anything I can do before reinstalling the operating system?

jmcerrejon, 2022-10-17:
> My friends, if I install the driver and crash or no longer enter the operating system, is there anything I can do before reinstalling the operating system?

Just go to a Terminal (CTRL + ALT + F1 for example) and type what I post [here](https://github.com/jmcerrejon/PiKISS/issues/165#issuecomment-1279925030).

douglasdock, 2022-10-17:
Next weekend I will test it, thank you very much

bagong, 2022-10-17:
Could it somehow be related to locale settings? I did the following: setup a fresh Pi OS 64-bit, updated to the latest, installed build-essentials and then build the Vulkan-drivers 22.2 via piKiss. The desktop came up fine (4k 30Hz), vulkan-tools worked as expected. Then I reconfigured the locale via Pi-config, set it to German, and since then the desktop crashes. I can still ssh into the system, but bluetooth-mouse and keyboard are not connected.

darksaviorx, 2022-12-01:
22.3 is out.

douglasdock, 2022-12-09:
Goodnight
I tried to install the wulkan driver, but again it crashed on raspiberry IOS 64bits.
The installation was normal, however when I tried to use the graphics card and the system did not start after restarting.
For that, I'm installing the drivers from this link

https://qengineering.eu/install-vulkan-on-raspberry-pi.html

wulkan - 20.3

darksaviorx, 2023-03-08:
"The bugfix release 22.3.7 is now available. This is the last release of the 22.3 series. Users are encouraged to
switch to the 23.0 series to continue receiving bugfixes." https://lists.freedesktop.org/archives/mesa-dev/2023-March/225936.html

jmcerrejon, 2023-03-09:
> "The bugfix release 22.3.7 is now available. This is the last release of the 22.3 series. Users are encouraged to
> switch to the 23.0 series to continue receiving bugfixes." https://lists.freedesktop.org/archives/mesa-dev/2023-March/225936.html

Tested my script on RPi OS 64 bits updated with 22.3.7 and... It works! Please, check it out and thumps up on this comment to know If It works on your side. Thank you, guy!

jmcerrejon, 2023-03-11:
It works on RPi OS 64 bits (at least not get a black screen when reboot once installed). So I'll close the issue. Thanks to everyone for the help.

closed 2023-03-11 by jmcerrejon

darksaviorx, 2023-03-11:
I only use 64bit Bullseye Lite with RetroPie so that's my only test. 

The frontends RetroPie uses (Emulationstation and Pegasus-Frontend) crash after 22.3.7 is installed. Maybe it works better on the desktop os but I don't use it. I also tried 23.0.0 with the same results. At this point I give up on Vulkan on Bullseye. I'll wait for Bookworm.

jmcerrejon, 2023-03-14:
> I only use 64bit Bullseye Lite with RetroPie so that's my only test.
>
> The frontends RetroPie uses (Emulationstation and Pegasus-Frontend) crash after 22.3.7 is installed. Maybe it works better on [...]

Try the new settings on the script thanks to https://github.com/kleinerm.