jmcerrejon/PiKISS #167: Re-Volt RVGL for 64 bit

Jojo-A, 2022-10-18:
Hi,

thanks a ton for your great effort!
If I understand it correctly, PiKiss not yet supports the 64 bit version of rvgl.

I've got it running on my 64 bit OS by simply using the pre-compiled ODROID C2 packages from the user "meveric" from the ODROID community:
https://oph.mdrjr.net/meveric/pool/main/r/rvgl-data-odroid/rvgl-data-odroid_21.0905a-1_all.deb
https://oph.mdrjr.net/meveric/pool/main/r/rvgl-odroid/rvgl-odroid_21.0905a-1+deb11_arm64.deb

The order is important: first the rvgl-data, second the game. Of course the already stated dependencies are still required in advance.

I hope this maybe helps to add Re-Volt 64 bit to PiKiss :) .

Thank you and best wishes!

jmcerrejon, 2022-10-18:
Great addition!. Thanks for your collaboration 😃

Jojo-A, 2022-10-18:
> Great addition!. Thanks for your collaboration smiley

The minimum I could do :) .
To be honest I am still a big fan of the ODROIDs. Even the old devices have been great gaming devices long before the RPi has been.

I will test some more packages from the ODROID. Since rvgl worked without any problems I will definitely try Serious Sam as well. If you are interested I can share my results (for 64 bit though).

jmcerrejon, 2022-10-18:
Sure, thanks. I must confess I'm an ODROID fan, too (I own the C1 and X4). In fact, this board was supported by PiKISS at the beginning 😅

Jojo-A, 2022-10-18:
Haha, me too! C1, C2 and XU4 :sweat_smile: ! Now that all my ODROIDs are in use in other projects, the RPi 4 is all thats left ;) .
So maybe you are familiar with meverics repositories and his awesome effort in context with his OGST (Odroid GameStation Turbo) project. There might be even more packages that can be ported...

closed 2025-11-30 by jmcerrejon