jmcerrejon/PiKISS #170: Quake fps is too low

c-a-s-m, 2022-10-31:
Using 1920x1080 resolution, 24 bit color, 60Hz refresh rate, no vsync at the game start map timerefresh command shows 7 (seven) fps. It's too low for game of 1996.
640x480 gives 40 fps. It's very low too but minimally playable.
Is there a way to replace a provided quake fork with a different one, running much faster? I believe fps more than 150 is required for 1920x1080.

c-a-s-m, 2022-10-31:
I'm using TwisterOS 2.1.2.

jmcerrejon, 2022-11-01:
Hi!

Tested with no hassle on RPi OS 32 Bits updated using your settings. Probably something related to Twister OS. Anyway, this is an installer app, so It only installs the game 😅 Maybe you can find support at the project URL [here](https://sourceforge.net/projects/quakespasm).

Good Luck!

I let you with a video showing a gameplay I recorded right now.

https://user-images.githubusercontent.com/1942431/199211244-117af017-af99-4451-ae91-7652ed916b3f.mp4

closed 2022-11-01 by jmcerrejon

c-a-s-m, 2022-11-02:
Sorry, cannot playback your video.

I've installed Raspberry Pi OS (64-bit). It's a little better. 17 fps in 1920x1080 and 85 in 640x480. Again, it's not 150 requested.

jmcerrejon, 2022-11-02:
Let you in my last comment the official repository. Ask the developers there about the performance you want. Good luck!

c-a-s-m, 2022-11-02:
Raspberry Pi OS (32-bit). 1920x1080 - 30 fps, 640x480 - 147.

The link you provided mentions only PC versions. They don't make the version for Raspberry Pi. Why to ask them about the performance on Raspberry Pi?

The binaries the piKISS script downloads are from misapuntesde.com site. I doubt authors of Quakespasm put them there.

Again, I'm not saying that someone did something wrong. I want to understand the reason of such a low frame rate. Is Raspberry Pi CPU slow? I doubt. Is Raspberry Pi GPU slow? Maybe, how to test and compare it with something? Does the executable utilize GPU not in the perfect way (making more textures passes then GPU supports, for example)? I don't know, maybe the author of the executable will say? Is there a way to use software-only rendering? Maybe Vulkan rendering is better?

jmcerrejon, 2022-11-02:
The link I provided is from the creators of the engine. I downloaded & compiled the source code from there and I hosted it on my site, but... I can’t do anything more beyond that. Any issues, bad performance, etc… just talk with their developers. I’m not the guy you need, believe me. I pretend to be polite, but this is my last comment on this thread. hope you understand me and again...good luck with that.

c-a-s-m, 2022-11-04:
Looks like my expectations on Pi's GPU power are far from reality. Waiting for Pi5 then. ;)

jmcerrejon, 2022-11-04:
I’ve compiled the latest version 0.95 but again, I get on my RPI 4, 4 GB a great performance. Stay tuned this weekend and try it