jmcerrejon/PiKISS #171: OpenXCOM supports Terror from the Deep, the script does not try to install it

c-a-s-m, 2022-11-04:
The game content is expected to be installed into TFTD directory at the same level UFO one resides.

jmcerrejon, 2022-11-05:
I don't know If this is a question or a bug 😅
PiKISS doesn't install game content (check the README.md) protected by laws in some countries.
I'm compiling the latest code just in case, and it now appears 2 directories: TFTD & UFO. I guess now you can install both and choose the game you prefer (If you copy the game content from your original copy of the game, of course).

c-a-s-m, 2022-11-05:
Sorry, my bad. The ticket is not exactly correct. I'm saying about the "magic air copy" part of the script.

That part works only if the source is a pre-made archive and not a directory.
I cannot point magic-air-copy-pikiss.txt to a live installation of the game on my PC, for example. If it's /media/pi/EXTERNAL_DRIVE/games/UFO, where the directory contains
common
standard
TFTD
UFO
openxcom.exe
README.md
CHANGELOG.txt
LICENSE.txt
the script creates ~/games/openxcom/UFO, puts everything into it (we have ~/games/openxcom/UFO/UFO, ~/games/openxcom/UFO/TFTD and unneeded ~/games/openxcom/UFO/{common,standard,openxcom.exe,README.md,...}.

So, it does not create ~/games/openxcom/TFTD and puts wrong content info ~/games/openxcom/UFO.

That may be applicable to Heroes3 too, for example. I have to have a local installation of the game on my PC and http-accessible archive of Data&Maps&Mp3 as well.

I believe it's better if "installation from directory" takes only required files and puts them into proper places.

jmcerrejon, 2022-11-06:
Ok, so is working If you copy your game files into the game's directory manually, but not through a directory using _magic-air-copy-pikiss.txt_, right?

c-a-s-m, 2022-11-06:
1. It works if copying files manually,
2. It works if magic-air-copy takes files from a premade archive,
3. It does not work if magic-air-copy takes files from directory. First it takes more files than required, second it places them incorrectly.