jmcerrejon/PiKISS #173: RetroArch 1.11.1-1 it's not working

douglasdock, 2022-12-07:
RetroArch 1.11.1-1 it's not working.

The previous version (1.10.3) was functional, could check the bug in the new version (1.11.1-1.) or go back to version 1.10.3

I noticed that the Retroarch configuration file (retroarch.cfg) is missing

jmcerrejon, 2022-12-07:
Please, tell me what the issue is to reproduce it. Thanks.

douglasdock, 2022-12-07:
Good afternoon my friend, how are you?
I had the previous version and it was working normally.
I had to reinstall "Retroarch".
However, after installation the program does not enter.
Clicking on the RetroArch icon, simply nothing happens.

I thought it could be the video driver configured in the "retroarch.cfg"
file, however when clicking on the file there was only one (1) command line.
I searched the internet for the same file and copied the command lines from
the "retroarch.cfg" file, however it still wouldn't load.

The previous version (1.10.3) worked normally.

Thank you for your help

Rak1ta, 2022-12-24:
I tried to install the version for Debian Buster - everything works. The retroarch folder is present and the config too.
[retroarch.cfg.txt](https://github.com/jmcerrejon/PiKISS/files/10298349/retroarch.cfg.txt)
[retroarch_qt.cfg.txt](https://github.com/jmcerrejon/PiKISS/files/10298350/retroarch_qt.cfg.txt)

closed 2024-06-09 by jmcerrejon