jmcerrejon/PiKISS #175: Many missing dependencies during OpenMW install on Raspberry Pi OS 64 bit

alexanderi96, 2023-01-11:
`Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Package libqtgui4 is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'libqtgui4' has no installation candidate
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package libboost-filesystem1.67.0
E: Couldn't find any package by glob 'libboost-filesystem1.67.0'
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package libboost-program-options1.67.0
E: Couldn't find any package by glob 'libboost-program-options1.67.0'
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package libbullet2.87
E: Couldn't find any package by glob 'libbullet2.87'`

JedMeister, 2023-01-11:
Looks like the `libqtgui4` package (and it's dependencies) only exist in Debian 10/Buster - not in 11/Bullseye (current stable)?!

I had a quick look, but there is no obvious successor to that package in Bullseye (at least not obvious to me). Although I did notice that there is a "openmw' package in the [Debian Bullseye repo](https://packages.debian.org/bullseye/openmw) - it's not the latest release, but it may be good enough?

I.e. just try installing with apt:
```
apt update
apt install -y openmw
```

(please note that I'm not the dev nor at all involved in this project - just a random stranger passing through...)

alexanderi96, 2023-01-12:
It works, thanks :) unfortunately since it is an older version my savegames are not compatible but it is better than nothing!

JedMeister, 2023-01-12:
Glad that's better than nothing! :smile: A slightly newer version (0.47.0 vs 0.46.0) is currently in [testing](https://packages.debian.org/bookworm/openmw). I thought that it may have installed cleanly, but it appears not:
```
$ sudo apt install -y ./openmw_0.47.0-3_amd64.deb 
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Note, selecting 'openmw' instead of './openmw_0.47.0-3_amd64.deb'
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 openmw : Depends: openmw-data (= 0.47.0-3) but it is not going to be installed
          Depends: libavcodec59 (>= 7:5.0) but it is not installable
          Depends: libavformat59 (>= 7:5.0) but it is not installable
          Depends: libavutil57 (>= 7:5.0) but it is not installable
          Depends: libc6 (>= 2.34) but 2.31-13+deb11u5 is to be installed
          Depends: libmyguiengine3debian1v5 (>= 3.4.1) but it is not going to be installed
          Depends: librecast1 (>= 1.5.1+git20210215.e75adf8) but it is not installable
          Depends: libstdc++6 (>= 12) but 10.2.1-6 is to be installed
          Depends: libswresample4 (>= 7:5.0) but it is not installable
          Depends: libswscale6 (>= 7:5.0) but it is not installable
E: Unable to correct problems, you have held broken packages.
```

Adding testng repos to a stable system is a really bad idea, so I don't suggest that. If you were really keen, then the [OpenMW wiki](https://wiki.openmw.org/index.php?title=Development_Environment_Setup#Debian_and_Ubuntu) notes that there is an Ubuntu PPA that will allow you to install a newer version.

A word of warning though, adding Ubuntu PPAs to Debian based systems is also generally a bad idea! Whilst Ubuntu is Debian based, it's not binary compatible, and foreign packages can potentially cause issues. Having said that, as they explicitly note that in the section titled "Debian and Ubuntu" - perhaps it's been carefully done to ensure compatibility? I can't be sure and even if it does work initially, it still may cause issues in the future. At the very least, I recommend cloning your OS (e.g. if you're running off an SD card, copy it for safe keeping) so worst case you can roll back if it breaks stuff.