jmcerrejon/PiKISS #176: GTA3 without desktop

Exarkuniv, 2023-01-13:
im working on getting this going on RetroPie. since its more or less the same thing as the raspberry pi OS it should work. 

so far everthing else i have tryed by you as worked amasingly. 
and i know that GTA3 works. i installed it with a clean raspberry OS and it played great. 

but no matter what i have tried i cant get it work
i have it going through XINIT with this .sh to start 

#!/bin/bash
cd /home/pi/games/GTAIII && ./re3

and this is what i get
```
[DBG]: cdvd_stream: read info 0x2178c78
[DBG]: Using one streaming thread for all channels
[DBG]: Created cdstream thread
[DBG-2]: [../src/skel/glfw/glfw.cpp.psInitialize:484]: gGameState = GS_START_UP
casepath couldn't find dir/file "gta3.set", full path was gta3.set
[DBG-2]: Default skin set as no other skins are available OR saved skin not found!
[DBG]: Physical memory size 3838128128
[DBG]: Available physical memory 451665920
wm_set_cursor_visibility: visible using XCreateFontCursor
GLFW Error: GLX: Failed to create context: GLXBadFBConfig
OpenGL version: 2.1 Mesa 19.3.2
/home/pi/games/GTAIII/re3: symbol lookup error: /home/pi/games/GTAIII/re3: undefined symbol: glfwUpdateGamepadMappings
```
i also have tried matchbox-window-manager as it helped other games in the past.

what am i missing ?
any thought would be great

symbios24, 2023-01-28:
Hi try this,enable the kms driver instead the fkms

jmcerrejon, 2024-01-13:
Sorry, RetroPie is not supported for PiKISS. Maybe the best you can do is to go to the RetroPie issue channel and get there correct answer. Good luck!

closed 2024-01-13 by jmcerrejon