jmcerrejon/PiKISS #178: Installation Not Working

3301n, 2023-03-12:
I am getting the following output when i enter the installation script `curl -sSL https://git.io/JfAPE | bash` in my terminal

`curl: (35) Recv failure: Connection reset by peer`

jmcerrejon, 2023-03-12:
Are you using Raspberry Pi OS 32 or bits? Yesterday precisely I did a New OS installation and installed PiKISS with that command with no hassle

3301n, 2023-03-12:
Not using the raspberry os, but trying to install in another flavour of linux running on raspberry pi 64 bit. Is it possible? It would be grateful if you could send the instructions. Also make it for general not limited to the raspberry pi os only , a script that works on arm64 devices.

jmcerrejon, 2023-03-12:
PiKISS is only supported on RPiOS, sorry. Don't have enough time for giving support to another OS or devices

closed 2023-03-12 by jmcerrejon