jmcerrejon/PiKISS #180: piKiss netools don't work on pi3b+ with bullseye

johncblacker, 2023-03-28:
I installed piKiss then used it's netool install script.  But, when I try to run "/home/pi/sc/opensource/netool.sh I get errors indicating no such file or directory and it's pointing to the /root/opensource/ directory.  Several errors like:
"cat: /root/opensource/netool.sh: No such file or directory"
"cat: /root/opensource/toolkit_config: No such file or directory" - 13 of these messages.
I don't know if it's something I did or didn't do; I presumed the install script would create the necessary files and subdirectories that are necessary for correct functioning.  I need help with this, otherwise I think I'll just drop the whole piKiss application altogether.

hzykiss, 2024-01-08:
Hello, you have solved the problem. I encountered the same problem with CM4 8G. No matter what program is installed, the system prompts "No such file or directory" when running.

jmcerrejon, 2024-01-13:
Unfortunately, Nettools is not updated anymore, but Metasploit is actively maintained. I'll update my script to support the installation of Metasploit (but It's very easy to install by the way 😉).

jmcerrejon, 2024-01-13:
Fixed at da1feed4fc2326a14c876b654ddb53cd66427285

Thank you, guys!

closed 2024-01-13 by jmcerrejon