jmcerrejon/PiKISS #188: Forever saying "Please answer (y)es or (n)o."

Botspot, 2023-06-28:
```
$ curl -sSL https://git.io/JfAPE | bash
Please answer (y)es or (n)o.
Please answer (y)es or (n)o.
Please answer (y)es or (n)o.
Please answer (y)es or (n)o.
Please answer (y)es or (n)o.
Please answer (y)es or (n)o.
Please answer (y)es or (n)o.
Please answer (y)es or (n)o.
Please answer (y)es or (n)o.
Please answer (y)es or (n)o.
Please answer (y)es or (n)o.
Please answer (y)es or (n)o.
Please answer (y)es or (n)o.
Please answer (y)es or (n)o.
Please answer (y)es or (n)o.
Please answer (y)es or (n)o.
Please answer (y)es or (n)o.
Please answer (y)es or (n)o.
Please answer (y)es or (n)o.
Please answer (y)es or (n)o.
Please answer (y)es or (n)o.
Please answer (y)es or (n)o.
Please answer (y)es or (n)o.
Please answer (y)es or (n)o.
Please answer (y)es or (n)o.
Please answer (y)es or (n)o.
Please answer (y)es or (n)o.
Please answer (y)es or (n)o.
Please answer (y)es or (n)o.
Please answer (y)es or (n)o.
```
Run in a terminal on a system without dialog installed. How long has this been broken?

Botspot, 2023-06-28:
This bug also occurs when trying to install PiKISS from Pi-Apps.

Botspot, 2023-06-29:
More specifically, this error only happens when there is already a `~/piKiss` folder in existence, the `dialog` package is uninstalled, and you try to run the installation command. `curl -sSL https://git.io/JfAPE | bash`
I encountered this error on a new system where I had the piKiss folder in my home directory from an older install, then tried to install PiKISS. This would also happen if a user first clones your repo with git and then runs your installation script.

jmcerrejon, 2023-07-01:
Don't know how long this it happens, because I don't use Pi-Apps. I'll check it out, anyway.

Botspot, 2023-07-01:
It can be caused without pi-apps. It happens when you run the install command when a pikiss folder already exists in $HOME.

jmcerrejon, 2023-07-01:
Fixed 9b4d87a75ce5f3131304896076a7cc65e4557338

Thanks for the report! ☺️

closed 2023-07-01 by jmcerrejon